Freescale/meta-freescale #1852: Build failure for core-image-weston scarthgap

fabioaiuto, 2024-06-21:
My build fails on gstreamer1.0-plugins-bad on task do_configure:

| ../git/meson.build:1:0: ERROR: Compiler aarch64-fsl-linux-gcc -mcpu=cortex-a53+crc+crypto -mbranch-protection=standard -fstack-protector-strong -O2 -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/home/fabio/fsl-community-bsp-scarthgap/build/tmp/work/cortexa53-crypto-mx8mp-fsl-linux/gstreamer1.0-plugins-bad/1.22.5.imx/recipe-sysroot cannot compile programs.
| 
| A full log can be found at /home/fabio/fsl-community-bsp-scarthgap/build/tmp/work/cortexa53-crypto-mx8mp-fsl-linux/gstreamer1.0-plugins-bad/1.22.5.imx/build/meson-logs/meson-log.txt
| ERROR: meson failed
| WARNING: exit code 1 from a shell command.
ERROR: Task (/home/fabio/fsl-community-bsp-scarthgap/sources/meta-freescale/recipes-multimedia/gstreamer/gstreamer1.0-plugins-bad_1.22.5.imx.bb:do_configure) failed with exit code '1'
NOTE: Tasks Summary: Attempted 4993 tasks of which 4992 didn't need to be rerun and 1 failed.
NOTE: Writing buildhistory
NOTE: Writing buildhistory took: 1 seconds

Summary: 1 task failed:
  /home/fabio/fsl-community-bsp-scarthgap/sources/meta-freescale/recipes-multimedia/gstreamer/gstreamer1.0-plugins-bad_1.22.5.imx.bb:do_configure

I get the log as suggested and see:

-----
Sanity check compile stderr:
cc1: error: '-Wno-error=return-mismatch': no option '-Wreturn-mismatch'; did you mean '-Wargument-mismatch'?

Then checked recent 3d61da6c1828479b58bb443711c8129e70ebace0 commit fixing things for GCC-14 but my yocto build is using

Detecting compiler via: `aarch64-fsl-linux-gcc -mcpu=cortex-a53+crc+crypto -mbranch-protection=standard -fstack-protector-strong -O2 -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/home/fabio/fsl-community-bsp-scarthgap/build/tmp/work/cortexa53-crypto-mx8mp-fsl-linux/gstreamer1.0-plugins-bad/1.22.5.imx/recipe-sysroot --version` -> 0
stdout:
aarch64-fsl-linux-gcc (GCC) 13.3.0 <------------------- this is my version

What's best for me to do?

Thanks a lot in advance

fabio

fabioaiuto, 2024-06-21:
I just realized that gcc version is 13:
```
$ grep -r ^GCCVERSION
poky/meta/conf/distro/include/tcmode-default.inc:GCCVERSION ?= "13.%"
```
Is the only way to proceed updating gcc to 14?

Thanks a lot in advance,

fabio

MaxKrummenacher, 2024-06-21:
The fixes for GCC-14 are only needed in master.
I guess the way forward is to revert those which introduce build failures in scarthgap. (And keep the backports were the build succeeds.)
I will check the commits which were backported and target GCC-14 and send a MR.

closed 2024-06-21 by otavio

fabioaiuto, 2024-06-25:
I've successfully built a core-weston-image now.
Thank you @MaxKrummenacher and everyone involved

fabio

PR #1853 (2024-06-21): [Scarthgap] Revert backports which don't compile with GCC-13
Closes #1852