Freescale/meta-freescale #1859: qt6 qtbase bbappend is overwriting PACKAGECONFIG from meta-qt6

georgekelly1097, 2024-06-25:
On my Verdin IMX8 dev board, meta-freescale/dynamic-layers/qt6-layer/recipes-qt/qt6/qtbase_%.bbappend is overwriting PACKAGECONFIG_GRAPHICS. meta-qt6 puts some variables in there, namely "wayland". Because meta-freescale is overwriting this value, you can't build qt6 + wayland when using meta-freescale. 

Of course I can make a new recipe myself to add "wayland" back in, but shouldn't meta-freescale not overwrite this variable from meta-qt6?

thochstein, 2024-06-26:
Hi @georgekelly1097, wayland is provided here: https://github.com/Freescale/meta-freescale/commit/25d78fb6bde4be99d444ca3db7bb224227f05e09. I tested on `master` with `MACHINE=imx8qm-mek bitbake qtbase -g > bitbake.log` and found wayland is set.

georgekelly1097, 2024-06-26:
Ah ok. We were using a Toradex manifest that was pointing to meta-freescale d649a5de - committed just 2 days before the fix you mentioned! Updating meta-freescale does seem to fix the issue. Thanks.

closed 2024-06-26 by georgekelly1097