Freescale/meta-freescale #1861: Upgrade BSP to LF6.6.23_2.0.0

otavio, 2024-07-02:
The immediate items to work on are:

- [ ]  Upgrade Linux kernel and headers (i.MX variants)
- [ ]  Upgrade Linux kernel (FSLC forks i.MX variants)
- [ ]  Upgrade U-Boot (FSLC forks i.MX variants)
- [ ]  Upgrade OP-TEE
- [ ]  Upgrade ATF (i.MX variants)
- [ ]  Upgrade Multimedia Packages
- [ ]  Upgrade Graphics Packages
- [ ]  Upgrade Jailhouse
- [ ]  Upgrade mcore-demos
- [ ]  Upgrade firmware-imx*
- [ ]  Upgrade imx-mkimage

The Release Notes are [here](https://www.nxp.com/docs/en/release-note/IMX_LINUX_RELEASE_NOTES.pdf) for checking.

ricardosalveti, 2024-07-16:
@angolini @MrCry0 ^ :-)

hiagofranco, 2024-07-23:
Hi @otavio, are these statuses updated?
For example, I see linux-fslc was already updated and backported to scarthgap. Thanks!

hiagofranco, 2024-08-26:
Hi @otavio, now that this issue is closed, can we start the backports to Scarthgap? How does it work?

If that is the case, I am happy to open a PR with the backports, I am just asking first to understand how the developing is done. These backports is something we need into our BSP. Thank you!

otavio, 2024-08-26:
Sure, this is complete, indeed.

Specifically, regarding the backports, I think the best way to do that is to cherry-pick every commit that should be backported to the stable branch. Still, we can also use this opportunity to backport fixes from the master to the branch.

We must be careful not to import patches that break the branch missing changes from the master from the OpenEmbedded-Core.

PR #1906 (2024-08-23): optee: Upgrade optee recipes from 4.0.0.imx to 4.2.0.imx
Fixes #1861