Freescale/meta-freescale #1882: OpenCV support with OpenCL on imx8mp

meetgandhi-eic, 2024-07-29:
Hi,

I'm trying to enable OpenCL support in OpenCV.
I'm on `hardknott` branch.

With reference to this discussion #320 , I have added below parameters in my `local.conf` file:

```
PACKAGECONFIG_append_pn-opencv = " opencl"
PREFERRED_PROVIDER_opencl-headers_imxgpu     = "imx-gpu-viv"
```

I have build the `opencv` package. As per the yocto's `log.do_configure` the support for `OpenCL` is enabled:

```
--   OpenCL:                        YES (no extra features)
--     Include path:                /home/test/imx8mp-sr/imx-yocto/build-xwayland-imx8mpsolidrun/tmp/work/cortexa53-crypto-mx8mp-poky-linux/opencv/4.5.2.imx-r0/git/3rdparty/include/opencl/1.2
--     Link libraries:              Dynamic load
--
```

I have verified the same from the `opencv2/cvconfig.h` file as well:

```
/* OpenCL Support */
#define HAVE_OPENCL
/* #undef HAVE_OPENCL_STATIC */
/* #undef HAVE_OPENCL_SVM */
```

But when I try to check the OpenCV in the generated image, it says OpenCL is disabled.
Below is the output of `opencv_version --opencl` command:
```
4.5.2
OpenCL is disabled
```

I have also tried below code snippet:
```
#include <iostream>
#include <fstream>
#include <string>
#include <iterator>
#include <opencv2/cvconfig.h>
#include <opencv2/opencv.hpp>
#include <opencv2/core/ocl.hpp>

using namespace std;

int main()
{
    cv::ocl::setUseOpenCL(true);
    if (!cv::ocl::haveOpenCL())
    {
        cout << "OpenCL is not avaiable..." << endl;
    }
    cv::ocl::Context context;
    if (!context.create(cv::ocl::Device::TYPE_ALL))
    {
        cout << "Failed creating the context..." << endl;
    }

    // In OpenCV 3.0.0 beta, only a single device is detected.
    cout << context.ndevices() << " GPU devices are detected." << endl;
    for (int i = 0; i < context.ndevices(); i++)
    {
        cv::ocl::Device device = context.device(i);
        cout << "name                 : " << device.name() << endl;
        cout << "available            : " << device.available() << endl;
        cout << "imageSupport         : " << device.imageSupport() << endl;
        cout << "OpenCL_C_Version     : " << device.OpenCL_C_Version() << endl;
        cout << endl;
    }
    return 0;
}

```

I got below output:
```
OpenCL is not avaiable...
Failed creating the context...
0 GPU devices are detected.
```

I tried using the OpenCL SDK directly. Using that my test application was able to detect the GPUs:

```
#include <iostream>
#include <string>
#include <vector>

#include <CL/opencl.hpp>

static cl_int PrintPlatformInfoSummary(cl::Platform platform)
{
    std::cout << "\tName:           " << platform.getInfo<CL_PLATFORM_NAME>()
              << "\n";
    std::cout << "\tVendor:         " << platform.getInfo<CL_PLATFORM_VENDOR>()
              << "\n";
    std::cout << "\tDriver Version: " << platform.getInfo<CL_PLATFORM_VERSION>()
              << "\n";

    return CL_SUCCESS;
}

static void PrintDeviceType(const std::string& label, cl_device_type type)
{
    std::cout << label << ((type & CL_DEVICE_TYPE_DEFAULT) ? "DEFAULT " : "")
              << ((type & CL_DEVICE_TYPE_CPU) ? "CPU " : "")
              << ((type & CL_DEVICE_TYPE_GPU) ? "GPU " : "")
              << ((type & CL_DEVICE_TYPE_ACCELERATOR) ? "ACCELERATOR " : "")
              << ((type & CL_DEVICE_TYPE_CUSTOM) ? "CUSTOM " : "") << "\n";
}

static cl_int PrintDeviceInfoSummary(const std::vector<cl::Device> devices)
{
    for (size_t i = 0; i < devices.size(); i++)
    {
        std::cout << "Device[" << i << "]:\n";

        cl_device_type deviceType = devices[i].getInfo<CL_DEVICE_TYPE>();
        PrintDeviceType("\tType:           ", deviceType);

        std::cout << "\tName:           "
                  << devices[i].getInfo<CL_DEVICE_NAME>() << "\n";
        std::cout << "\tVendor:         "
                  << devices[i].getInfo<CL_DEVICE_VENDOR>() << "\n";
        std::cout << "\tDevice Version: "
                  << devices[i].getInfo<CL_DEVICE_VERSION>() << "\n";
        std::cout << "\tDevice Profile: "
                  << devices[i].getInfo<CL_DEVICE_PROFILE>() << "\n";
        std::cout << "\tDriver Version: "
                  << devices[i].getInfo<CL_DRIVER_VERSION>() << "\n";
    }

    return CL_SUCCESS;
}

int main(int, char**)
{
    std::vector<cl::Platform> platforms;
    cl::Platform::get(&platforms);
    std::cout << "Enumerated " << platforms.size() << " platforms.\n\n";

    for (size_t i = 0; i < platforms.size(); i++)
    {
        std::cout << "Platform[" << i << "]:\n";
        PrintPlatformInfoSummary(platforms[i]);

        std::vector<cl::Device> devices;
        platforms[i].getDevices(CL_DEVICE_TYPE_ALL, &devices);

        PrintDeviceInfoSummary(devices);
        std::cout << "\n";
    }

    std::cout << "Done.\n";

    return 0;
}

```

The output is shown below:

```
Enumerated 1 platforms.

Platform[0]:
	Name:           Vivante OpenCL Platform
	Vendor:         Vivante Corporation
	Driver Version: OpenCL 3.0 V6.4.3.p2.336687
Device[0]:
	Type:           GPU 
	Name:           Vivante OpenCL Device GC7000UL.6204.0000
	Vendor:         Vivante Corporation
	Device Version: OpenCL 3.0 
	Device Profile: FULL_PROFILE
	Driver Version: OpenCL 3.0 V6.4.3.p2.336687
Device[1]:
	Type:           GPU 
	Name:           Vivante OpenCL Device VIP8000Nano-S+I.8002.0000
	Vendor:         Vivante Corporation
	Device Version: OpenCL 3.0 
	Device Profile: FULL_PROFILE
	Driver Version: OpenCL 3.0 V6.4.3.p2.336687
Done
```

Am I missing something here, when building the OpenCV with OpenCL?
Let me know if you need any additional information.

fabioestevam, 2024-08-08:
Hardknott is no longer supported. Please try scarthgap instead.

meetgandhi-eic, 2024-08-10:
Upgrading the entire Yocto version isn’t feasible at the moment. Could you suggest some alternative solutions or workarounds?

otavio, 2024-08-10:
As Fabio pointed out, it is very difficult for us to assist you with such an old version. Unfortunately, this is not maintained anymore, as it has reached the end of life. So, upgrading the release is likely to become imperative very soon.

To work on that kind of situation, we usually make some specific backports or minor modifications to allow the customers to keep using the old releases. However, this ends up becoming more expensive and, in the long term, unmaintainable.

We can help you commercially if you want, but our advice is to try to Upgrade the version as fast as possible.