Freescale/meta-freescale #1887: firmware-imx: DP-Firmware communication times out since 8.22

tq-steina, 2024-07-30:
Hi,

I noticed that since firmware-imx_8.22 the `dpfw.bin` DP-Firmware for i.MX8 times out during intial communication. My platform is TQMa8x featuring an i.MX8QM and a DP output.. In my case the firmware itself is already loaded by u-boot, resulting in the following kernel messages:
```
[    2.409104] cdns-mhdp-imx 56268000.hdmi: lane-mapping 0x1b
[    2.414614] cdns-mhdp-imx 56268000.hdmi: Failed to get i2c-over-aux-retries - using default 5
[    2.423157] cdns-mhdp-imx 56268000.hdmi: i2c-over-aux-retries 0x05
[    2.431904] cdns-mhdp-imx 56268000.hdmi: Direct firmware load for dpfw.bin failed with error -2
[    2.439401] [drm] Started firmware!
[    2.440630] cdns-mhdp-imx 56268000.hdmi: Falling back to sysfs fallback for: dpfw.bin
[    2.456416] [drm] set maximum defer retry to 5
[    2.456433] [drm] HDP FW Version - ver 35083 verlib 20560
[    2.479050] [drm] Only HDCP 2.2 is enabled
[    7.483304] cdns-mhdp-imx 56268000.hdmi: [drm:cdns_mhdp_reg_read] *ERROR* Failed to read register.
```
This log is from v6.6, but it's also happening for v6.1.
Once I revert the following commits on top top of `scarthgap` branch on meta-freescale DP works as expected:
* b346c0544023cb22650fd9ab131de52f58edef2e
* b346c0544023cb22650fd9ab131de52f58edef2e
* 8d9953331d407a5f713052f8891b134b114d9dce
* 8d9953331d407a5f713052f8891b134b114d9dce
* d3a380dd892ff2429d40db3b09ff2c6686a909a8
* d3a380dd892ff2429d40db3b09ff2c6686a909a8
* 56454b14d1ac9e709ad1d15962c7653458f78997
* 56454b14d1ac9e709ad1d15962c7653458f78997

Here is the log from firmware-imx_8.18:
```
[    2.384569] cdns-mhdp-imx 56268000.hdmi: lane-mapping 0x1b
[    2.390084] cdns-mhdp-imx 56268000.hdmi: Failed to get i2c-over-aux-retries - using default 5
[    2.398642] cdns-mhdp-imx 56268000.hdmi: i2c-over-aux-retries 0x05
[    2.407872] cdns-mhdp-imx 56268000.hdmi: Direct firmware load for dpfw.bin failed with error -2
[    2.412061] [drm] Started firmware!
[    2.416605] cdns-mhdp-imx 56268000.hdmi: Falling back to sysfs fallback for: dpfw.bin
[    2.432896] [drm] set maximum defer retry to 5
[    2.432906] [drm] HDP FW Version - ver 34559 verlib 20560
[    2.452048] [drm] Only HDCP 2.2 is enabled
[    2.465703] cdns-mhdp-imx 56268000.hdmi: wrong link rate index
[    2.475780] imx-drm display-subsystem: bound 56268000.hdmi (ops cdns_mhdp_imx_ops)
[    2.484179] [drm] Initialized imx-drm 1.0.0 20120507 for display-subsystem on minor 0
```

I noticed the version difference `34559` <-> `35083`, but I have no information what actually changed. This is FW upgrade require an additional change which I do not know?

otavio, 2024-08-24:
Is it possible for you to check with the latest release?

tq-steina, 2024-09-11:
You mean firmware-imx_8.24? I did, that's when I noticed the problem. As far as I can tell, it's the same firmware version for `dpfw.bin`, according to Linux kernel log.

tq-steina, 2025-01-07:
Some update here. Using firmware-imx_8.26 things seems to work again as expected on a 6.6.23 kernel:
```
[    2.361994] cdns-mhdp-imx 56268000.hdmi: lane-mapping 0x1b
[    2.367498] cdns-mhdp-imx 56268000.hdmi: Failed to get i2c-over-aux-retries - using default 5
[    2.376042] cdns-mhdp-imx 56268000.hdmi: i2c-over-aux-retries 0x05
[    2.385240] [drm] Started firmware!
[    2.388842] cdns-mhdp-imx 56268000.hdmi: Direct firmware load for dpfw.bin failed with error -2
[    2.397562] cdns-mhdp-imx 56268000.hdmi: Falling back to sysfs fallback for: dpfw.bin
[    2.398629] [drm] set maximum defer retry to 5
[    2.405413] [drm] HDP FW Version - ver 35083 verlib 20560
[    2.425250] [drm] Only HDCP 2.2 is enabled
[    2.438652] cdns-mhdp-imx 56268000.hdmi: wrong link rate index
[    2.448750] imx-drm display-subsystem: bound 56268000.hdmi (ops cdns_mhdp_imx_ops)
[    2.457059] [drm] Initialized imx-drm 1.0.0 20120507 for display-subsystem on minor 0
```

otavio, 2025-01-07:
Is there anything pending on this task or should we close this?

tq-steina, 2025-01-07:
For now nothing is pending. Thanks

closed 2025-01-07 by tq-steina