Freescale/meta-freescale #1928: Update linux-fslc to latest mainline release?

Livius90, 2024-09-08:
@otavio 
What is the roadmap/timeplan to update linux-fslc recipe to use the latest Linux kernel version?

I tried to use linux-fslc v6.6.36 for i.MX93 EVK board and it seems to me many important things is missing in the device-tree source of imx93 and in pmic driver (NXP pca9451a). Latest mainline got that many things, just need to update to use [v6.10.9](https://www.kernel.org/), i guess.

otavio, 2024-09-09:
For the update, we need people to step up to help. I'm not against to maintain the latest stable version but someone needs to help sending updates for the main repository as well as the update for here to update the recipe.

Are you willing to help?

Livius90, 2024-09-09:
Do you have any custom patches which need to apply to the latest stable kernel version later, all the time? What is the fork source of linux-fslc?

I can test a normal build for v6.10.9 and i can test booting and the basics in i.MX7D and i.MX93 at the moment.

Livius90, 2024-09-10:
@otavio 
Linux kernel **v6.10.9 from mainline** is ready for meta-freescale scarthgap.
I tested the build in my forks ([linux-fslc](https://github.com/Livius90/linux-fslc/tree/6.10.x%2Bfslc), [meta-freescale](https://github.com/Livius90/meta-freescale/tree/scarthgap)), it works without any error, following was needed to get it:
https://github.com/Freescale/meta-freescale/compare/scarthgap...Livius90:meta-freescale:scarthgap

In next days i will test it, how can work/boot on an i.MX93 EVK board.

Livius90, 2024-09-11:
@otavio 
[linux-fslc v6.10.9](https://github.com/Livius90/linux-fslc/tree/6.10.x%2Bfslc) -> i.MX7D pico pi :white_check_mark: work in [scarthgap](https://github.com/Livius90/meta-freescale/tree/scarthgap)
[linux-fslc v6.10.9](https://github.com/Livius90/linux-fslc/tree/6.10.x%2Bfslc) -> i.MX93 11x11 EVK :white_check_mark: [scarthgap](https://github.com/Livius90/meta-freescale/tree/scarthgap)

Livius90, 2024-09-12:
@otavio 
https://github.com/Freescale/meta-freescale/pull/1939

otavio, 2024-09-15:
As I mentioned in the pull request that you sent, I'm totally ok in going to the next update. I would also like to thank you for the work in this update as well.

closed 2024-09-28 by Livius90