Freescale/meta-freescale #1947: Error when building gstreamer1.0-plugins-good_1.24.0.imx

TheNageek, 2024-09-17:
The exact error is shown below, 

```c++
../git/ext/qt6/gstqsg6material.cc:32:10: fatal error: 'private/qrhi_p.h' file not found
| #include <private/qrhi_p.h>
|          ^~~~~~~~~~~~~~~~~~
| 1 error generated.
```

I also tested [this GStreamer patch](https://gitlab.freedesktop.org/gstreamer/gstreamer/-/merge_requests/7131/diffs), but still having the error.

```c++
../git/ext/qt6/gstqsg6material.cc:32:10: fatal error: 'QtGui/private/qrhi_p.h' file not found
| #include <QtGui/private/qrhi_p.h>
|          ^~~~~~~~~~~~~~~~~~~~~~~~
| 1 error generated.
```

hiagofranco, 2024-09-20:
Hi @TheNageek, do you have any details of your build to share? We have plugins-good building with Scarthgap without any issues

TheNageek, 2024-09-20:
Hey @hiagofranco, thanks for getting back to me, currently we're on on `gstreamer1.0_1.22.0.imx` and I'm attempting to backport `gstreamer1.0_1.24.0.imx`, we'd like to use some of the `webrtc` elements of 1.24.0.

TheNageek, 2024-09-20:
Yes, that's what we were hoping to do.

hiagofranco, 2024-09-20:
Ah ok, so the issue text is a bit misleading, it is not true the recipe fails in scarthgap.

But apart from that, you should backport all the gstreamer recipes to the same version 1.24.0. If you are trying to backport only the plugins-good, it will likely fail. Gstreamer recommends having all the plugins with the same version, so you could try to do that and check if it works... I think even if you backport everything you might find some errors, the gstreamer 1.24.0 is supposed to be used with scarthgap only (which is the newer NXP BSP).

I hope this helps ;-)

TheNageek, 2024-09-20:
That's true, it's not completely scarthgap meta-freescale, apologies there.  But in regards to the 1.24.0, I just copy-pasted gstreamer contents from scarthgap.  Do you think a better option would be to backport all of meta-freescale scarthgap?

otavio, 2024-09-21:
I do believe that upgrading to the latest release is best. The hassle of maintaining the old version is bigger than you can guess.

Normally, the upgrade path is very easy and pays off in the longer term, especially for the latest LTS, which will be maintained for the next four years. 

Maintaining those Frankstsain systems usually takes too much time, and the effort to make workarounds is not worth it.

Either way, we can't fix something here, so I'm closing this issue as it belongs to your site.

closed 2024-09-21 by otavio