Freescale/meta-freescale #2033: Update linux-fslc to latest kernel v6.12 LTS?

Livius90, 2024-11-20:
Hi,

Linux kernel v6.12 released some days ago, it seems it will be the next LTS release. Do you support to update it in meta-freescale from v6.11 to v6.12? I am most interested to start using the new Real-Time options for ARM64 i.MX SoCs.

https://www.developer-tech.com/news/linux-kernel-6-12-real-time-capabilities-hardware-boosts-and-more/

angolini, 2024-11-21:
Could you, please, prepare a PR?

Livius90, 2024-11-21:
> Could you, please, prepare a PR?

For it need to make a new `6.12.x+fslc` branch in [linux-fslc](https://github.com/Freescale/linux-fslc).

Livius90, 2024-11-24:
```
git clone --single-branch --branch linux-6.12.y https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git
cd linux/
git branch -m linux-6.12.y 6.12.x+fslc
git remote add fslc-mirror https://github.com/Freescale/linux-fslc.git
git push -u fslc-mirror
```

otavio, 2024-11-25:
I have created the branch so you can prepare a pull request now. Sorry for taking longer to create it for you to work.

Livius90, 2024-11-30:
PR was done and build works for me.

closed 2024-11-30 by Livius90