Freescale/meta-freescale #2048: Upgrade BSP to LF6.6.52_2.2.0

otavio, 2024-12-16:
The immediate items to work on are:

- [ ]  Upgrade Linux kernel and headers (i.MX variants)
- [ ]  Upgrade Linux kernel (FSLC forks i.MX variants)
- [ ]  Upgrade U-Boot (FSLC forks i.MX variants)
- [ ]  Upgrade OP-TEE (i.MX variants)
- [ ]  Upgrade ATF (i.MX variants)
- [ ]  Upgrade Multimedia Packages
- [ ]  Upgrade Graphics Packages
- [ ]  Upgrade Jailhouse
- [ ]  Upgrade mcore-demos
- [ ]  Upgrade firmware-imx*
- [ ]  Upgrade imx-mkimage

The Release Notes are [here](https://www.nxp.com/docs/en/release-note/IMX_LINUX_RELEASE_NOTES.pdf) for checking.

lms-jk, 2025-01-16:
[OID_registry.patch](https://github.com/Freescale/meta-freescale/blob/5c0779f683f6ef8ddf83fd8d4cdc65fb13f32227/recipes-kernel/linux/linux-imx/0001-lib-build_OID_registry-don-t-mention-the-full-path-o.patch) can no longer be applied to the updated sources:
```
DEBUG: Executing python function extend_recipe_sysroot
NOTE: Direct dependencies are ['/home/jkircher/yocto/build/../layers/poky/meta/recipes-devtools/quilt/quilt-native_0.68.bb:do_populate_sysroot', '/home/jkircher/yocto/build/../layers/poky/meta/recipes-kernel/kern-tools/kern-tools-native_git.bb:do_populate_sysroot', 'virtual:native:/home/jkircher/yocto/build/../layers/poky/meta/recipes-devtools/patch/patch_2.7.6.bb:do_populate_sysroot']
NOTE: Installed into sysroot: ['quilt-native', 'patch-native']
NOTE: Skipping as already exists in sysroot: ['gettext-minimal-native', 'cmake-native', 'libtool-native', 'texinfo-dummy-native', 'kern-tools-native', 'openssl-native', 'expat-native', 'ncurses-native', 'zlib-native', 'git-native', 'make-native', 'perl-native', 'bzip2-native', 'xz-native', 'zstd-native', 'attr-native', 'ca-certificates-native', 'curl-native', 'gdbm-native']
DEBUG: sed -e 's:^[^/]*/:/home/jkircher/yocto/build/tmp/work/phyboard_pollux_imx8mp_2-lms-linux/linux-imx/6.6.52+git/recipe-sysroot-native/:g' /home/jkircher/yocto/build/tmp/sysroots-components/x86_64/quilt-native/fixmepath | xargs sed -i -e 's:FIXMESTAGINGDIRTARGET:/home/jkircher/yocto/build/tmp/work/phyboard_pollux_imx8mp_2-lms-linux/linux-imx/6.6.52+git/recipe-sysroot:g; s:FIXMESTAGINGDIRHOST:/home/jkircher/yocto/build/tmp/work/phyboard_pollux_imx8mp_2-lms-linux/linux-imx/6.6.52+git/recipe-sysroot-native:g' -e 's:FIXME_PSEUDO_SYSROOT:/home/jkircher/yocto/build/tmp/sysroots-components/x86_64/pseudo-native:g' -e 's:FIXME_HOSTTOOLS_DIR:/home/jkircher/yocto/build/tmp/hosttools:g' -e 's:FIXME_PKGDATA_DIR:/home/jkircher/yocto/build/tmp/pkgdata/phyboard-pollux-imx8mp-2:g' -e 's:FIXME_PSEUDO_LOCALSTATEDIR:/home/jkircher/yocto/build/tmp/work/phyboard_pollux_imx8mp_2-lms-linux/linux-imx/6.6.52+git/pseudo/:g' -e 's:FIXME_LOGFIFO:/home/jkircher/yocto/build/tmp/work/phyboard_pollux_imx8mp_2-lms-linux/linux-imx/6.6.52+git/temp/fifo.110996:g'
DEBUG: Python function extend_recipe_sysroot finished
DEBUG: Executing shell function do_patch
(1/3) 0001-tty-vt-conmakehash-Don-t-mention-the-full-path-of-th.patch
(2/3) 0001-lib-build_OID_registry-don-t-mention-the-full-path-o.patch
[INFO]: check of .kernel-meta//patches//./0001-lib-build_OID_registry-don-t-mention-the-full-path-o.patch with "git am" did not pass, trying reduced context.
[INFO]: Context reduced git-am of .kernel-meta//patches//./0001-lib-build_OID_registry-don-t-mention-the-full-path-o.patch with "git am" did not work, trying "apply".
error: patch failed: lib/build_OID_registry:8
error: lib/build_OID_registry: patch does not apply
[ERROR]: Application of .kernel-meta//patches//./0001-lib-build_OID_registry-don-t-mention-the-full-path-o.patch failed.
         Patch needs to be refreshed. Sample resolution script:
             .git/rebase-apply/resolve_rejects
ERROR: Could not apply patches for phyboard-pollux-imx8mp-2.
ERROR: Patch failures can be resolved in the linux source directory /home/jkircher/yocto/build/tmp/work/phyboard_pollux_imx8mp_2-lms-linux/linux-imx/6.6.52+git/git)
WARNING: exit code 1 from a shell command.
```

Oquirella, 2025-03-21:
There is already a new release and this issue is still not completed...

angolini, 2025-03-24:
I could not find the bump PR for optee, for me this is the only missing piece.

@thochstein do you have plan to work on this?

thochstein, 2025-03-24:
I hope so in the next few days.

closed 2025-03-27 by otavio

Oquirella, 2025-03-28:
This should also be ported to scartgaph

PR #2068 (2025-01-21): Add imxviv and imxmali overrides & upgrade to LF6.6.52_2.2.0
Add imxviv and imxmali machine overrides to the model to easily differentiate between Vivante and Mali configuration.

Upgrade graphics to LF6.6.52_2.2.0. Fixes: #2048.