Freescale/meta-freescale #2051: CVEs affecting ffmpeg 4.4.1

colin-pm, 2025-01-02:
Hello,

After running cve-check on the Kirkstone branch, several CVEs have been identified with ffmpeg 4.4.1.

- CVE-2022-48434
- CVE-2023-46407
- CVE-2023-47470
- CVE-2024-7272
- CVE-2024-22860
- CVE-2024-22862

I've experimented with applying the patches for these CVEs to ffmpeg 4.4.1. All of the patches have merge conflicts. Four of the CVE patches do not even appear to apply to files that exist in 4.4.1, meaning the CVE might not exist on 4.4.1, or is hidden somewhere else in the code. Upgrading ffmpeg might be the better solution, but 1c6c0f6 indicates there is a blocker from being able to upgrade ffmpeg. Will this be resolved so a newer version of ffmpeg can be used?

otavio, 2025-01-02:
I don't expect this going to be worked in the Kirkstone branch, as there is a new version already included in new Scarthgap release.

colin-pm, 2025-01-03:
This appears to also affect the newest release as well, which also includes ffmpeg 4.4.1.

colin-pm, 2025-06-05:
There appear to be a new batch of CVEs that affect ffmpeg 4.4.1.

- [CVE-2024-31578](https://nvd.nist.gov/vuln/detail/CVE-2024-31578)
- [CVE-2024-36617](https://nvd.nist.gov/vuln/detail/CVE-2024-36617)
- [CVE-2024-7055](https://nvd.nist.gov/vuln/detail/CVE-2024-7055)
- [CVE-2025-1373](https://nvd.nist.gov/vuln/detail/CVE-2025-1373)
- [CVE-2025-1594](https://nvd.nist.gov/vuln/detail/CVE-2025-1594)