Freescale/meta-freescale #2078: Build regression against Styhead

prevas-emkan, 2025-01-23:
Hi,

After commit 1e3228f1da57 ("imx-atf: Adjust for compiler virtual renames in oe-core"), the master branch is no longer compatible with Styhead. In meta-poky, commits

- 6b1e65b7d6ce ("base: Switch virtual/cross-XXX to be under recipe specific providers")
- 90e0a0f7f453 ("classes/recipes: Switch virtual/XXX-gcc to virtual/cross-cc (and c++/binutils)")

are in master and _not_ styhead and thus meta-freescale's master branch is no longer compatible with styhead.

angolini, 2025-01-27:
I'm confused why it's a regression and not normal development.
Would you mind preparing a PR for fixing the compatibility?

Villemoes, 2025-01-28:
There are two problems. One on master, which could be fixed by a PR dropping the claim of styhead compatibility.

The other is the _lack_ of a styhead branch; that cannot be created by a PR, so someone with push access should create a styhead branch based on the last commit which was actually compatible with styhead, d615badf . That branch should probably then also gain a commit dropping the claimed walnascar compat, but that again can't be done as a PR until the styhead branch actually exists.

prevas-emkan, 2025-01-28:
> I'm confused why it's a regression and not normal development.

I mean a regression in the sense that this layer claims to be compatible with styhead via `LAYERSERIES_COMPAT`, which is no longer true after commit `1e3228f1da57 ("imx-atf: Adjust for compiler virtual renames in oe-core")`.

> Would you mind preparing a PR for fixing the compatibility?

I can create a PR to drop styhead from `LAYERSERIES_COMPAT`, but then none of the current branches works with styhead. Someone should create a styhead branch from the latest compatible commit with styhead, `d615badf93ec`, which I don't have access to.

Villemoes, 2025-01-28:
We will automatically test this tonight, but we do have a nightly build from about a week ago when master was exactly at that commit which succeeded, so now that we can point our nightly builds at the styhead branch rather than master, it should work again. I'll report back tomorrow.

Villemoes, 2025-01-29:
As expected, the nightly build succeeded. I've created two follow-up PRs (#2085 , #2086 ) adjusting the compatibility claimed on styhead and master.

This issue can be closed, but I'm not allowed to use the "Close with comment" button.

closed 2025-01-29 by angolini