Freescale/meta-freescale #2100: UBOOT_ENTRYPOINT for mx93

joggee-fr, 2025-01-30:
Hi,

I am working on i.MX 93 EVK and wanted to test FIT image to be loaded by U-Boot.
The `scarthgap` branch is also used but after investigations I am pretty sure the issue is same on `master`.
A custom machine has been created but it is mainly just an include of `imx93-11x11-lpddr4x-evk.conf`.

Adding the following lines allows to generate the FIT image including Linux kernel and device trees.
```
KERNEL_CLASSES += "kernel-fitimage"
KERNEL_IMAGETYPES += "fitImage"
```

However, looking at the generated image, the kernel load address does not seem to be correct.
```
$ dumpimage -l ./fitImage
FIT description: Kernel fitImage
Created:         Tue Oct 22 14:04:24 2024
 Image 0 (kernel-1)
  Description:  Linux kernel
  Created:      Tue Oct 22 14:04:24 2024
  Type:         Kernel Image
  Compression:  gzip compressed
  Data Size:    14929159 Bytes = 14579.26 KiB = 14.24 MiB
  Architecture: AArch64
  OS:           Linux
  Load Address: 0x01314c40
  Entry Point:  0x01314c40
  Hash algo:    sha256
  Hash value:   cc7d05f42b8e27e265fbfbdfb315496561d53fb8650dbe748514849f0bd2f41f
```

This is due to `UBOOT_ENTRYPOINT` not defined. In fact, this is exactly the same behavior than issue #791 but this time for mx93.

Setting `UBOOT_ENTRYPOINT = "0x80400000"` in custom machine configuration is a good workaround.

Maybe it could be interesting to directly add it here. PR #1414 does not add a `UBOOT_ENTRYPOINT` definition. But it seems the following line could be added to `imx-base.inc`.

```
UBOOT_ENTRYPOINT:mx93-generic-bsp   ?= "0x80400000"
```

Am I right?

otavio, 2025-01-30:
Yes, that would be a perfect fix to solve this issue for many people. Are you willing to prepare a pull request to fix that?

joggee-fr, 2025-01-30:
Thanks @otavio, I have opened PR #2104 for this topic.

closed 2025-01-31 by otavio