Freescale/meta-freescale #2128: When updating to the latest verrsion of the kirkstone branch my bitbake gives the following error.

tbandtg, 2025-02-13:
Initialising tasks: 100% |#############################################################################################| Time: 0:00:02
Sstate summary: Wanted 192 Local 0 Mirrors 0 Missed 192 Current 3184 (0% match, 94% complete)
Removing 88 stale sstate objects for arch host: 100% |#########################################################| Time: 0:00:00
Removing 26 stale sstate objects for arch armv8a: 100% |###############################################################| Time: 0:00:00
Removing 3 stale sstate objects for arch x86_64: 100% |################################################################| Time: 0:00:00
NOTE: Executing Tasks
ERROR: imx-boot-1.0-r0 do_deploy: ExecutionError('/mnt/build/mhost2/build/tmp/work/-poky-linux/imx-boot/1.0-r0/temp/run.do_deploy.882348', 1, None, None)
ERROR: Logfile of failure stored in: /mnt/build/mhost2/build/tmp/work/-poky-linux/imx-boot/1.0-r0/temp/log.do_deploy.882348
Log data follows:
| DEBUG: Executing python function sstate_task_prefunc
| DEBUG: Python function sstate_task_prefunc finished
| DEBUG: Executing shell function do_deploy
| Set boot target as flash_evk
| cp: '/mnt/build/tony/mhost2/build/tmp/work/-poky-linux/imx-boot/1.0-r0/deploy-imx-boot/imx-boot' and '/mnt/build/tony/mhost2/build/tmp/work/-poky-linux/imx-boot/1.0-r0/deploy-imx-boot/imx-boot' are the same file
| WARNING: exit code 1 from a shell command.
ERROR: Task (/mnt/build/mhost2/sources/meta-imx/meta-bsp/recipes-bsp/imx-mkimage/imx-boot_1.0.bb:do_deploy) failed with exit code '1'
NOTE: Tasks Summary: Attempted 7962 tasks of which 7911 didn't need to be rerun and 1 failed.
NOTE: Writing buildhistory
NOTE: Writing buildhistory took: 3 seconds

angolini, 2025-02-17:
Can you please, give more details on the instructions to reproduce the issue?

tbandtg, 2025-02-18:
sure
<manifest>

  
  <default sync-j="2"/>

  
  <remote fetch="https://github.com/openembedded" name="oe"/>
  <remote fetch="https://git.yoctoproject.org/git" name="yocto"/>
  <remote fetch="https://github.com/Freescale" name="community"/>
  <remote fetch="https://github.com/nxp-imx" name="nxp-imx" />
  <remote fetch="https://github.com/kraj" name="clang"/>
  <remote fetch="https://github.com/OSSystems" name="OSSystems"/>
  <remote fetch="https://git.yoctoproject.org" name="security"/>
  <remote fetch="https://github.com/intel-iot-devkit" name="cloud"/>
  <remote fetch="https://github.com/ADVANTECH-Corp" name="meta-advantech2" />
  <remote fetch="https://code.qt.io/yocto" name="QT6" />
  <remote fetch="https://github.com/TimesysGit" name="Timesys" />
  <remote fetch="https://github.com/nxp-imx-support" name="imx-support" />
  <remote fetch="git@bitbucket.org:company/company-yocto-meta-layers.git/src/dev-adva/" name="meta-company"/>
  <remote fetch="https://github.com/Igalia" name="webkit" />

  
  
  
  
  <project name="fsl-community-bsp-base" path="sources/base" remote="community" revision="60f79f7af60537146298560079ae603260f0bd14" upstream="kirkstone">
  <linkfile src="README" dest="README"/>
  </project>
  
  
  
  
  
  <project name="meta-advantech2" path="sources/meta-advantech" remote="meta-advantech2" revision="ce92107d835064318e1cda259124c0cfde84531e" upstream="aim34-kirkstone" dest-branch="aim34-kirkstone">
  </project>
  
  
  
  
  
  
  <project name="meta-browser" path="sources/meta-browser" remote="OSSystems" revision="d5542fc35528bfadaafd67f109033434788e550b"/>
  
  
   
  
  <project name="meta-clang" path="sources/meta-clang" remote="clang" revision="fabeeadc6a1cbf1a81fbdb7dc9fa4022c7e5e851" upstream="kirkstone"/>
  
  
  
  
  
  <project name="meta-freescale" path="sources/meta-freescale" remote="community" revision="c525e0c19bdc46d45f71873b5f286f49abb69418" upstream="kirkstone" />
  
  
  
  
  
  <project name="meta-freescale-3rdparty" path="sources/meta-freescale-3rdparty" remote="community" revision="48516272d97f9415e0ee3ab54a923e4c40143e69" upstream="kirkstone"/>
  
  
    
  
  
  <project name="meta-freescale-distro" path="sources/meta-freescale-distro" remote="community" revision="d5bbb487b2816dfc74984a78b67f7361ce404253" upstream="kirkstone"/>
    
  
  
    
  
  
  <project name="meta-webkit" path="sources/meta-webkit" remote="webkit" revision="4a49b63860100ad0b7d9050448dc4023364bfbfd" upstream="kirkstone"/>
    
  
  
  <project name="meta-imx" path="sources/meta-imx" remote="nxp-imx" revision="refs/tags/rel_imx_5.15.71_2.2.2" upstream="kirkstone-5.15.71-2.2.2" >
  
  <linkfile dest="README-IMXBSP" src="README"/>
  </project>
  
  
  
  
  <project name="meta-arm" remote="yocto" path="sources/meta-arm" revision="936c02ec13661bd86a05f7f90e1b920d5092d670" upstream="kirkstone"/>
  
  
  
  
  
  <project name="meta-openembedded" path="sources/meta-openembedded" remote="oe" revision="de8681b4a2a101b99dd2c48d89a7de2ccd9a961f" upstream="kirkstone"/>
  
  
  
  
  
  <project name="meta-timesys" path="sources/meta-timesys" remote="Timesys" revision="c2f65e99c8ac3001452a2c436a4a22413d2f95ef" upstream="kirkstone"/>
  
  
  
  
  
  <project name="poky" path="sources/poky" remote="yocto" revision="3272e64a731280206d13fe746e4b9e3a5e7c73a8" upstream="kirkstone"/>
  
  
  
  
    <project name="meta-qt6" path="sources/meta-qt6" remote="QT6" revision="75a2511884de8494fff93fba0354f5542be41cdf" upstream="6.3"/>
  
  
  
  
  <project name="meta-security" path="sources/meta-security" remote="security" revision="06c240c3c235fa06fe724e278d53463ac67cc26b" upstream="kirkstone"/>
  
  
  
  
  <project name="meta-iot-cloud" path="sources/meta-iot-cloud" remote="cloud" revision="77764cbf6d1486d48018426d6c7de49c9ddcdc25" upstream="kirkstone"/>
  
  
  <project name="meta-lts-mixins" path="sources/meta-lts-mixins" remote="security" revision="6b3a208f7e14138728a6e581ac75e8973ab48ef3" upstream="kirkstone/rust"/>
 
  
 <project name="meta-company" path="sources/meta-company" remote="meta-company" revision="Tony/upg/YMB-149-host-yocto-investigate-upgrading-chromium">
    <copyfile src="imx-setup-release.sh" dest="company-imx-setup-release.sh"/>
    <copyfile src="setup-environment" dest="company-setup-environment"/>
 </project>
</manifest>

find . -name imx-boot*
./meta-imx/meta-cockpit/recipes-bsp/imx-mkimage/imx-boot_%.bbappend
./meta-imx/meta-bsp/recipes-bsp/imx-mkimage/imx-boot_1.0.bbappend
./meta-imx/meta-bsp/recipes-bsp/imx-mkimage/imx-boot_1.0.bb
./meta-freescale/recipes-bsp/imx-mkimage/imx-boot_1.0.bb
./meta-freescale/classes/imx-boot-container.bbclass
./meta-advantech/meta-fsl-imx/recipes-bsp/imx-mkimage/imx-boot_%.bbappend

angolini, 2025-02-18:
We don't support this environment.

The way you introduced the log made me think you were reporting a bug in kirkstone, but it looks like you have an unique environment.

Let me try to detail how you could proceed from here.

### Option 1
You can try to reproduce using the supported env/machine (as described at http://freescale.github.io/), on master or scarthgap.

### Option 2
You can try to debug why you are facing the error

```
| cp: '/mnt/build/tony/mhost2/build/tmp/work/-poky-linux/imx-boot/1.0-r0/deploy-imx-boot/imx-boot' and '/mnt/build/tony/mhost2/build/tmp/work/-poky-linux/imx-boot/1.0-r0/deploy-imx-boot/imx-boot' are the same file
```

open the scripts from `temp` for `do_deploy` and see what is being running

closed 2025-02-24 by angolini