Freescale/meta-freescale #2208: imx-vpu-hantro-daemon out of sync with the kernel

landswellsong, 2025-04-24:
I'm investigating the errors I get while trying to encode video on IMX8MP with V4L2 layer and I get the encoder params all messed up (width=100, height=0, unsupported format which I added a print for and it's way out of range):
```
# cat /home/vsi/daemon.log
src/vsi_dec.c:176:vsi_dec_get_codec_format() INFO core_num 2, ReadAsicCoreCount 2
src/vsi_main.c:365:main() INFO main thread id = ffffbe525720
src/vsi_main.c:585:inst_init() DEBUG inst_init begin
src/vsi_main.c:626:inst_init() DEBUG inst_init ffffbe0df1a0 end
src/vsi_main.c:515:worker_threads() LOG worker_threads inst[2800000001]: codec_fmt 1, cmd id 1
src/vsi_main.c:515:worker_threads() LOG worker_threads inst[2800000001]: codec_fmt 1, cmd id 1
src/vsi_main.c:515:worker_threads() LOG worker_threads inst[2800000001]: codec_fmt 1, cmd id 1
src/vsi_main.c:515:worker_threads() LOG worker_threads inst[2800000001]: codec_fmt 1, cmd id 1
src/vsi_main.c:515:worker_threads() LOG worker_threads inst[2800000001]: codec_fmt 1, cmd id 1
src/vsi_main.c:515:worker_threads() LOG worker_threads inst[2800000001]: codec_fmt 1, cmd id 1
src/vsi_main.c:515:worker_threads() LOG worker_threads inst[2800000001]: codec_fmt 1, cmd id 1
src/vsi_main.c:515:worker_threads() LOG worker_threads inst[2800000001]: codec_fmt 1, cmd id 1
src/vsi_main.c:515:worker_threads() LOG worker_threads inst[2800000001]: codec_fmt 1, cmd id 0
src/vsi_enc.c:261:enc_cmd_processor() INFO Has Param when streamon.
src/vsi_enc_video_h2.c:1241:getAlignedPicSizebyFormat() ERROR not support this format
src/vsi_enc_video_h2.c:1275:get_config_from_cmd_h2() ERROR Invalid parameters width=100, height=0
src/vsi_enc.c:378:enc_state_processor() ERROR set_parameter() failed, ret=-101
src/vsi_main.c:515:worker_threads() LOG worker_threads inst[2800000001]: codec_fmt 1, cmd id 1
src/vsi_enc.c:365:enc_state_processor() INFO No source buffer @ INIT state, wait...
src/vsi_main.c:515:worker_threads() LOG worker_threads inst[2800000001]: codec_fmt 1, cmd id 16
src/vsi_main.c:515:worker_threads() LOG worker_threads inst[2800000001]: codec_fmt 1, cmd id 15
src/vsi_main.c:515:worker_threads() LOG worker_threads inst[2800000001]: codec_fmt 1, cmd id 3
src/vsi_main.c:528:worker_threads() DEBUG Instance [2800000001] codec finished all jobs
src/vsi_main.c:537:worker_threads() DEBUG worker_threads finish, instance id 2800000001
src/vsi_main.c:656:inst_destroy() DEBUG inst_destroy begin
src/vsi_main.c:685:inst_destroy() DEBUG inst_destroy finish
```

After a bunch of probing I've narrowed the issue down to the incompatibility between the contents of the `v4l2_daemon_enc_buffers` structure.

Kernel version at https://github.com/Freescale/linux-fslc/commit/5ff4cf4d61e11f0fdf8d4e2e54fbb203e46d34b2
```
struct v4l2_daemon_enc_buffers {
	/*IO*/
	s32 inbufidx;	 //from v4l2 driver, don't modify it
	s32 outbufidx;	 //-1:invalid, other:valid.

	u64 busLuma;
	s32 busLumaSize;
	u64 busChromaU;
	s32 busChromaUSize;
	u64 busChromaV;
	s32 busChromaVSize;

	u64 busLumaOrig;
	u64 busChromaUOrig;
	u64 busChromaVOrig;

	u64 busOutBuf;
	u32 outBufSize;

	u32 bytesused;	//valid bytes in buffer from user app.
	s64 timestamp;
	u32 average_qp;
};
```

Daemon version at 1.1.9:
```
typedef struct v4l2_daemon_enc_buffers {
  /*IO*/
  int32_t inbufidx;   // from v4l2 driver, don't modify it
  int32_t outbufidx;  //-1:invalid, other:valid.

  vpu_addr_t busLuma;
  int32_t busLumaSize;
  vpu_addr_t busChromaU;
  int32_t busChromaUSize;
  vpu_addr_t busChromaV;
  int32_t busChromaVSize;

  vpu_addr_t busLumaOrig;
  vpu_addr_t busChromaUOrig;
  vpu_addr_t busChromaVOrig;

  vpu_addr_t busOutBuf;
  uint32_t outBufSize;

  uint32_t bytesused;  // valid bytes in buffer from user app.
  int64_t timestamp;
} v4l2_daemon_enc_buffers;
```

There is a `average_qp` field introduced in https://github.com/Freescale/linux-fslc/commit/de80eb0b4d452f51bb441970e8bacae6fcdb81b6 and it seems to be shifting everything. I'll try patching it up for myself, but on a larger scale I assumed that the changes should be mirrored in a new hantro daemon version, but I don't seem to be able to find it anywhere and this repo's top version is also 1.1.9. 

If it works I'll do a PR with a patch, although will need some hand-holding to ensure the patch only applies for a daemon after a certain commit on the kernel if this makes any sense? Otherwise any input on whether there is a new daemon would be appreciated.

landswellsong, 2025-04-24:
Apparently there's a [1.2.0 out](https://www.nxp.com/lgfiles/NMG/MAD/YOCTO/imx-vpu-hantro-daemon-1.2.0-67fa7a7.tar.gz) (at least), found via [Digi's recipes](https://github.com/digi-embedded/meta-digi/blob/scarthgap/meta-digi-arm/dynamic-layers/freescale-layer/recipes-bsp/imx-vpu-hantro/imx-vpu-hantro-daemon_1.2.0.bb).

Since it includes a sha, I don't think if I can probe for any other versions and I'm unsure how this works with NXP overall, is there a public list of files on the Yocto mirrors?

There aren't much relevant changes so I assume there is an even newer version:
```Patch
diff --color -u -r imx-vpu-hantro-daemon-1.1.9/.scmversion imx-vpu-hantro-daemon-1.2.0-67fa7a7/.scmversion
--- imx-vpu-hantro-daemon-1.1.9/.scmversion     2023-10-11 13:47:34.000000000 +0300
+++ imx-vpu-hantro-daemon-1.2.0-67fa7a7/.scmversion     2024-05-06 13:32:05.000000000 +0300
@@ -1 +1 @@
-master+g82faef7
\ No newline at end of file
+MM_04.09.00_2405_L6.6.y+g67fa7a7
\ No newline at end of file
diff --color -u -r imx-vpu-hantro-daemon-1.1.9/v4l2_vsi_daemon/src/vsi_dec.c imx-vpu-hantro-daemon-1.2.0-67fa7a7/v4l2_vsi_daemon/src/vsi_dec.c
--- imx-vpu-hantro-daemon-1.1.9/v4l2_vsi_daemon/src/vsi_dec.c   2023-10-11 13:13:43.000000000 +0300
+++ imx-vpu-hantro-daemon-1.2.0-67fa7a7/v4l2_vsi_daemon/src/vsi_dec.c   2024-05-06 13:30:52.000000000 +0300
@@ -1620,7 +1620,10 @@
   //    ASSERT(h->dec_cur_state == DAEMON_DEC_STATE_DECODE);
 
   // Not flush all queued buffers on OUTPUT, according to spec4.5.1.10
-  h->dec_output_state = DAEMON_QUEUE_STATE_PEND_OFF;
+  if (h->dec_output_state == DAEMON_QUEUE_STATE_PAUSE)
+    h->dec_output_state2 = 1;
+  else
+    h->dec_output_state = DAEMON_QUEUE_STATE_PEND_OFF;
   h->dec_output_act = DAEMON_QUEUE_ACT_EOS;
   return 0;
 }
@@ -1665,6 +1668,11 @@
   ASSERT(h->dec_cur_state == DAEMON_DEC_STATE_DRAIN);
 
   h->dec_capture_state = DAEMON_QUEUE_STATE_ON;
+  if (h->dec_output_state == DAEMON_QUEUE_STATE_PAUSE) {
+    h->dec_output_state = DAEMON_QUEUE_STATE_ON;
+    if (h->dec_output_state2) h->dec_output_state = DAEMON_QUEUE_STATE_PEND_OFF;
+    h->dec_output_state2 = 0;
+  }
   h->dec_wait_dbp_buf = 0;
   return 0;
 }
@@ -1738,10 +1746,11 @@
     h->dec_output_state2 = 0;
   }
 
-  if (h->existed_dpb_nums || !h->seq_hdr_parsed)
+  if ((h->existed_dpb_nums && h->dec_capture_state == DAEMON_QUEUE_STATE_ON) || !h->seq_hdr_parsed)
     h->dec_output_state = DAEMON_QUEUE_STATE_ON;
   else
     h->dec_output_state = DAEMON_QUEUE_STATE_PAUSE;
+
   return 0;
 }
```

landswellsong, 2025-04-24:
Hmm `meta-imx` has [1.5.0](https://github.com/nxp-imx/meta-imx/blob/styhead-6.12.3-1.0.0/meta-imx-bsp/recipes-bsp/imx-vpu-hantro/imx-vpu-hantro-daemon_1.5.0.bb).

And it does have the relevant bits although I've not tested it yet all together.

otavio, 2025-04-24:
Exactly, that was exactly what I was going to send you just before you commented. It would be possible for you to prepare a pull request to update those recipes, so we are up to date?

landswellsong, 2025-04-24:
I'll do that tomorrow, I just want to verify if it fixes the underlying issue as well.

closed 2025-04-25 by otavio