Freescale/meta-freescale #2217: Default asound.conf and 44100 samplerate

ventureresearch, 2025-04-25:
The default asound.conf routes through this section for playback by default:
```
pcm.asymed{
type asym
playback.pcm "dmix_44100"
capture.pcm "dsnoop_44100"
}
```

In testing we were having audio skips, stutters, and dropouts during high CPU usage playing Youtube videos in Chromium.

Changing the default samplerate to 48000 to match the Youtube audio encoding resulted in no skipping or stutter, perhaps avoiding internal resampling 48000->44100 done by Chromium. 

Not sure if it would be better to change things here in meta-freescale, or if there is some setting or bug in Chromium that is preferring it to resample audio itself instead of opening up ALSA and thus the hardware PCM at 48000 directly. Or maybe Chromium just assumes a lot of extra CPU is available for resampling audio.

New:
recipes-bsp/alsa-state/alsa-state/imx-nxp-bsp/asound.conf
recipes-bsp/alsa-state/alsa-state/imx-mainline-bsp/asound.conf
```
pcm.asymed{
type asym
playback.pcm "dmix_48000"
capture.pcm "dsnoop_48000"
}
```

Anyways, just wanted to drop a note somewhere about having this issue, for others to find in the future using Chromium. Since you are providing the default asound.conf, I am reporting it here.

In current testing we are using IMX8M Plus, Scarthgap branches, kernel 6.6.52, chromium 117.

otavio, 2025-04-28:
I also don't see any reason to not do the change.

ventureresearch, 2025-04-28:
ok, submitting a PR. accidentally hit close on this...

closed 2025-04-28 by ventureresearch

thochstein, 2025-04-28:
Fine with me. I'll forward this on internally, but otherwise we'll just pick it up and see if it breaks any of our tests.