Freescale/meta-freescale #2384: imx-gpu-viv OpenVX and NN packages

andy9a9, 2025-09-25:
Dear community!

I would like to ask you if there is a correct way to split/not installing all `libOvx` and `libNN` libraries. As you can see they are quite big.
```
cortexa53-crypto-mx8mp-oe-linux/imx-gpu-viv/6.4.11.p2.8-aarch64/image/usr/lib# du -ha . | sort -rh
159M	.
57M	./libNNVXCBinary-evis2.so
17M	./libVSC.so
17M	./libNNGPUBinary-ulite.so
17M	./libNNGPUBinary-evis2.so
12M	./libOvx12VXCBinary-evis2.so
5.6M	./libOvxGPUVXCBinary-ulite.so
5.6M	./libOvxGPUVXCBinary-nano.so
5.6M	./libOvxGPUVXCBinary-lite.so
5.6M	./libOvxGPUVXCBinary-evis.so
3.5M	./libOpenVX.so.1.3.0
2.9M	./libSPIRV_viv.so
2.5M	./libCLC.so
2.1M	./libGAL.so
1.8M	./libGLESv2.so.2.0.0
1.3M	./libvulkan_VSI.so.1.3.3
1.3M	./libGLSLC.so
824K	./libOpenCL.so.3.0.0
392K	./libOpenCL.so.1.2.0
392K	./libGLESv1_CM.so.1.1.0
392K	./libGLESv1_CL.so.1.1.0
392K	./libGLES_CM.so.1.1.0
392K	./libGLES_CL.so.1.1.0
324K	./libOpenVG.3d.so.1.1.0
280K	./libEGL.so.1.5.0
212K	./libNNArchPerf.so
212K	./libArchModelSw.so
68K	./libVDK.so.1.2.0
68K	./libOpenVXU.so
68K	./libgbm_viv.so
68K	./libgbm.so.1.0.0
28K	./pkgconfig
```
The `imx-gpu-viv` recipe has basically a runtime dependency.
```bitbake
FILES:libopenvx-imx = " \
    ${libdir}/libOpenVX${REALSOLIBS} \
    ${libdir}/libOpenVXC${SOLIBS} \
    ${libdir}/libOpenVXU${SOLIBS} \
    ${libdir}/libOvx*${SOLIBS} \
    ${libdir}/libArchModelSw${SOLIBS} \
"
FILES:libopenvx-imx-dev = "${includedir}/VX ${libdir}/libOpenVX${SOLIBSDEV}"
RDEPENDS:libopenvx-imx = "libnn-imx ${OPENCL_VX_INTRINSIC_EXTENSION_PACKAGES}"

FILES:libnn-imx = "${libdir}/libNN*${SOLIBS}"
```
and if you're using the `OpenVX`, then you get all libraries. If we look at the list of available packages
```
cortexa53-crypto-mx8mp-oe-linux/imx-gpu-viv/6.4.11.p2.8-aarch64/packages-split$ ls
imx-gpu-viv                  libgal-imx.shlibdeps    libopencl-imx.shlibdeps
imx-gpu-viv-dbg              libgbm-imx              libopenvg-imx
imx-gpu-viv-demos            libgbm-imx-dev          libopenvg-imx-dev
imx-gpu-viv-demos.shlibdeps  libgbm-imx.shlibdeps    libopenvg-imx.shlibdeps
imx-gpu-viv-dev              libgles1-imx            libopenvx-imx
imx-gpu-viv-doc              libgles1-imx-dev        libopenvx-imx-dev
imx-gpu-viv-locale           libgles1-imx.shlibdeps  libopenvx-imx.shlibdeps
imx-gpu-viv-src              libgles2-imx            libspirv-imx
imx-gpu-viv-staticdev        libgles2-imx-dev        libspirv-imx.shlibdeps
imx-gpu-viv-tools            libgles2-imx.shlibdeps  libvdk-imx
imx-gpu-viv-tools.shlibdeps  libgles3-imx-dev        libvdk-imx-dev
libclc-imx                   libglslc-imx            libvdk-imx.shlibdeps
libclc-imx.shlibdeps         libglslc-imx-dev        libvsc-imx
libegl-imx                   libglslc-imx.shlibdeps  libvsc-imx.shlibdeps
libegl-imx-dev               libnn-imx               libvulkan-imx
libegl-imx.shlibdeps         libnn-imx.shlibdeps     libvulkan-imx-dev
libgal-imx                   libopencl-imx           libvulkan-imx.shlibdeps
libgal-imx-dev               libopencl-imx-dev
```
`libnn-imx` has no `-dev` variant, so can't say which library/symbol is matching with which header.

Would it be possible to create a correct dependencies, split the files into packages or shrink the libraries a little bit? It's little bit a problem if you wanted to use just a small part from the library and you will get `~20MB` library even if it's stripped.

Thank you

andy9a9, 2025-09-25:
> Looking in the original repo, that [dependency was added to fix an OpenVX CTS test case](https://github.com/nxp-imx/meta-imx/commit/acd8a9d97510e2568365b51d94ad22ee98d04cc5). Presumably, the dependency should have been added to the OpenVX CTS recipe.
> It looks like just dropping the dependency is the correct thing to do. Can you make a PR?

Sure, but I think, I didn't understand it. Which one is the `OpenVX CTS recipe`? The `imx-gpu-viv-6` is providing both variants, `libopenvx` and `libnn` too. Do you think about [meta-oe](https://github.com/openembedded/meta-openembedded/blob/master/meta-oe/recipes-graphics/vk-gl-cts/khronos-cts.inc) variant, or?
Or do you mean
```patch
- RDEPENDS:libopenvx-imx = "libnn-imx ${OPENCL_VX_INTRINSIC_EXTENSION_PACKAGES}"
+ RDEPENDS:libopenvx-imx = "${OPENCL_VX_INTRINSIC_EXTENSION_PACKAGES}"
```
something like this?

On the other hand, that's just a half of my request... . If we look at the files in `libnn-imx` package
```
cortexa53-crypto-mx8mp-oe-linux/imx-gpu-viv/6.4.11.p2.8-aarch64/packages-split/libnn-imx/usr/lib$ du -ha . | sort -hr
90M	.
57M	./libNNVXCBinary-evis2.so
17M	./libNNGPUBinary-ulite.so
17M	./libNNGPUBinary-evis2.so
212K	./libNNArchPerf.so
```
there are multiple libraries which are huge. Are they all needed to be installed? How to distinguish between them.

thochstein, 2025-09-25:
I made a fix removing the existing runtime dependency. Still waiting for information regarding splitting `libnn-imx`.

thochstein, 2025-10-18:
Hi @andy9a9, I asked the team internally and am waiting for a response.

thochstein, 2025-11-06:
Sorry, took me a while to get back to this. It turns out we do already have the SOC-specific packaging.

Still waiting for the team to respond regarding further optimization.